nirenjan/libx52 #5: 12 hour clock display broken for 12 AM

nirenjan, 2015-12-05:
When the user sets the clock to a time that represents 12:00:00 AM to 12:59:59 AM in the chosen location (local/GMT), and the clock is set to display in 12 Hour mode, the MFD displays 00:<mm> AM, i.e. the hour is displayed as 0.

This appears to be a known hardware issue, at least with my unit. Setting the hour value to 12 corresponds to 12 PM, so that won't fix the problem.

nirenjan, 2015-12-05:
This seems to be a problem even on clock 2 & clock 3 - 12 AM is displayed as 00 AM.

rogerxxxx, 2015-12-05:
Do you know of anybody that actually makes use of the MFD?

If so, the only method I can think of, is when there is a scenario where the computer keyboard is hidden from the user.  (ie. Arcade game or setup similar to a real simulation.)

Mange, 2015-12-05:
I, for one, use it.

To see the time. 😉

closed 2018-02-22 by nirenjan