nirenjan/libx52 #8: error: cannot install 'libx52util.la' to a directory not ending in /usr/local/lib

rogerxxxx, 2015-12-10:
./autogen && ./configure --prefix=/home/roger/src/x52pro-linux/install && make && make install

I get the following errors during "make install"

libtool: warning: remember to run 'libtool --finish /usr/local/lib'
libtool: warning: '../libx52/libx52.la' has not been installed in '/usr/local/lib'
libtool: warning: '../libx52/libx52.la' has not been installed in '/usr/local/lib'
libtool:   error: error: cannot install 'libx52util.la' to a directory not ending in /usr/local/lib

Looks like the previous is defined within:
./libx52/libx52.la:libdir='/usr/local/lib'
./util/libx52util.la:libdir='/usr/local/lib'

nirenjan, 2015-12-11:
I just tried and couldn't reproduce the problem. Can you try to run `git clean -xfd` from the root folder and try the steps again? There's a possibility that if there was any old content leftover by mistake, it could cause the issue you were seeing.

Based on http://stackoverflow.com/questions/10279829/installing-glib-in-non-standard-prefix-fails and https://bz.apache.org/bugzilla/show_bug.cgi?id=35098#c9, a `make clean` is always required after running configure, even on a fresh download. `git clean -xfd` just takes it one step further and deletes all files that are not part of the repo, including the ones ignored by `.gitignore`.

nirenjan, 2015-12-11:
I've also enabled Travis CI on the repo, so any failures should be caught early. It does

    ./autogen.sh && ./configure && make && make distcheck

Current status is [image]

rogerxxxx, 2015-12-11:
Thanks for the note concerning "git clean -xfd".

However, I already did previously perform a rm -rf ./x52pro-linux, which removed the root .git folders.

I'll redo everything again and see what happens.  Gentoo does not use /usr/local unless those packages are binary only or proprietary, nor do I use /usr/local.  Usually when I see this, it's usually because of the developer's custom settings are still preserved within the main repository.  Also note, I was not installing the package within my root /usr or /usr/local, and specified a local or $HOME folder for installing.

nirenjan, 2015-12-11:
I don't believe I have anything custom within the repo, and to the best of my knowledge, `make distcheck` does try to do an install in a non standard location, and that's passing.

Do let me know if you still have issues, and I'll try to bring up a Gentoo VM and test it there.

rogerxxxx, 2015-12-11:
Was still in the process of finessing my last comment. ;-)

rogerxxxx, 2015-12-11:
We can probably close this if those warnings are of no concern.

nirenjan, 2015-12-11:
I'm still kind of learning the Autotools/Libtool framework, and it seems like it's only the *.la files, which are only used by libtool and not needed after the files are successfully installed. Refer http://osdir.com/ml/db.monetdb.bugs/2008-02/msg00222.html.

closed 2015-12-11 by nirenjan

rogerxxxx, 2015-12-11:
Yea, I've read the book(s) too, and simply just understand the need and what is being performed.  But still find the tools extremely confusing, especially when debugging.  I also understand C structs and pointers better now, but typedefs seem like just another curve ball.

But I think we can pretty much conclude, you're doing tens times better than I, with the exception I have so little time at most times of the day.

I can see things rapidly changing, and on my latest compile, I see one remaining warning.

libtool: warning: relinking 'libx52util.la'

Cheers.  Looks really clean and ready for packaging.

nirenjan, 2015-12-11:
That last warning seems to be expected behavior based on http://osdir.com/ml/libtool-gnu/2009-12/msg00069.html

Rtoax, 2021-12-02:
when i `make install`, i got 
```
make[3]: Entering directory '/home/rongtao/rpm-4.17.0/lib'
make[4]: Entering directory '/home/rongtao/rpm-4.17.0/lib'
make[4]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/usr/lib64'
 /bin/sh ../libtool   --mode=install /usr/bin/install -c   librpm.la '/usr/lib64'
libtool:   error: error: cannot install 'librpm.la' to a directory not ending in /usr/local/lib
```

nirenjan, 2021-12-02:
@Rtoax, this doesn't look like my repository, but for what it's worth, did you try running it from a clean workspace?