nirenjan/libx52 #9: Compiling for macOS

ryandrake08, 2016-10-06:
Excellent library! Despite having -linux in its name, I found this software to work beautifully on macOS as well with a single change: Apple does not have ELIBACC, so you need to ./configure with -DELIBACC=EINVAL (or whatever error code you want to use that macOS supports). Other than that, it works! Feel free to conditionalize the use of ELIBACC on Linux or don't bother since the workaround is trivial.

Mange, 2016-10-07:
That's good to know! 😄 
A pull requests on the README detailing how to do this would be most welcome so other people can discover this more easily.

If you understand the code, you might even be able to implement the suggested change yourself. I don't dare touch code that enters the kernels, so I won't do it myself. 😉
I'd be willing to accept a PR for it, if the diff is small enough to be understandable for me so I can do a "faux review" of it.

Mange, 2016-10-07:
(Don't mind me. I didn't look at the repo name and accidentally though this was on another project that I maintain. It was completely unrelated to this repo and I apologize for it. Comment has been removed.)

nirenjan, 2016-10-07:
@ryandrake08, would you be willing to raise a PR for this? I haven't really tried to run this on OSX, so any help would be appreciated.

ryandrake08, 2016-10-08:
Please review: https://github.com/nirenjan/x52pro-linux/pull/10

closed 2016-10-08 by ryandrake08