nirenjan/libx52 #11: Support for x52 (non pro)

winiciuscota, 2016-12-22:
Hi, Does this driver work on the x52 (non pro)?

I just need a way to turn off those annoying leds

apollothethird, 2016-12-22:
I only have the pro.  It works very well and is easy to use.  Some one with the x52 (non pro) might comment, but I would suggest that you consider running a test and posting your results for others.

I think it's a good question and am interested.  Personally I would expect for most of the common functions between the two to actually work.

-- L. James

-- 
L. D. James
ljames@apollo3.com
www.apollo3.com/~ljames

nirenjan, 2016-12-22:
I have a check during initialization for the product code that is specific to the X52 Pro. Since I don't have the non-Pro version, I have no clue about the functionality. It should be fairly straightforward to add a case for the non-Pro model here - https://github.com/nirenjan/x52pro-linux/blob/master/libx52/x52_core.c#L30

I'd suggest along similar lines as @apollothethird to pull the source code, modify it to accept the non-pro version and run your tests.

winiciuscota, 2016-12-23:
Thanks for the instructions. I'll definitely try that when I have more time

nirenjan, 2017-01-11:
@winiciuscota, I've added support for the X52, but as the commit message states, you cannot update the LEDs individually. I'd need to know the control messages for those in order to update the library. However, you should be able to reduce the brightness down to zero for the LEDs, assuming that the control messages are the same between the pro and non-pro models.

winiciuscota, 2017-03-07:
@nirenjan Sorry for taking this long to test it. That worked perfectly!

nirenjan, 2017-03-07:
Could you try to verify if the other commands work? My assumption is that apart from setting the LED states, everything else should work as is.

winiciuscota, 2017-03-08:
All the commands are working(except for led and I didn't tested raw). Attempting to use led gives this message: Error: Operation not supported. The mfd command is awesome, I didn't even know it was possible to change the mfd text on the x52 non pro

nirenjan, 2017-03-08:
That's awesome! I'd still like to identify the LEDs, so let me know if you're ready to try some raw commands to try and turn on or off the LEDs.

winiciuscota, 2017-03-08:
I can do that as long as there is no risk of destroying my device

nirenjan, 2017-03-10:
Seems like the API is more or less the same between the X52 and X52 Pro. The only info I have though is that the LEDs are probably single color blue LEDs on the X52, while they are red/green LED pairs on the Pro.

The risk is minimal, IMHO.

nirenjan, 2017-03-10:
Can you describe what LEDs are glowing on the X52 when you first plug it in?

nirenjan, 2017-03-28:
The easiest way to add support for the LEDs would be to snoop the USB messages from a Windows instance which has the Saitek drivers installed. The manual is no help - it simply states that you can control the LED brightness, but doesn't say anything about the individual LEDs.

rogerxxxx, 2017-03-28:
Use wireshark, select the USB device for capturing packets, while only capturing when changing the color.  I started doing this myself, but have been quite busy.

nirenjan, 2020-05-06:
Well, according to [this website](https://forums.x-plane.org/index.php?/files/file/35304-x52luaout-winmaclin/), the LEDs cannot be individually controlled on the X52 (non-pro). Given that the library already supports everything else, I'm going to close this issue.

closed 2020-05-06 by nirenjan