nirenjan/libx52 #14: Browser support

nigeltiany, 2018-08-09:
Wondering if this controller is available under navigator.gamepads on any browser with this driver. Can someone with a linux pc and the controller please try it out with any of these websites? 

http://html5gamepad.com/
https://gamepadviewer.com/

rogerxxxx, 2018-08-09:
If the "navigator gamepad" web interface API simply checks the Linux joystick driver status for each axis or button action, then the default Linux joystick driver should work fine.

I believe this driver is a more customized driver for this Saitek device,  allowing user access to the more unnecessary features of the joystick such as LED lighting brightness and colors, as well as LED display.  

Regardless, web API should have access.  I would check to see if your web API is windows only, or Linux compatible.  If it is Linux compatible, I would take a gamble and buy the device with only worrying next if the throttle mapping (or other axis or switch) mapping isn't working correctly.  Most API's screw-up the Saitek mode switch, but again usually not a needed function.  I would worry more about the throttle axis working properly, as well as most needed buttons/axis.  For all I know, the joystick may work flawlessly.

(Personally I just desire or almost need the LED brightness settings, as the default brightness is much too bright.  Choosing colors is nice to help remember button mapping, but would imagine users rarely look downwards after some time.)

nirenjan, 2018-08-09:
Roger is right - the driver doesn't replace the standard Linux HID driver, but adds user-space APIs to control the MFD and LEDs, which aren't managed by standard USB.

nirenjan, 2018-08-09:
It shows up just fine with https://html5gamepad.com, although it's missing a couple of axes - notably the slider and mouse stick on the throttle. All the buttons seem to show up fine though.

With https://gamepadviewer.com, it detects the joystick, but I wasn't able to get the inputs to map to the XBox controller.

rogerxxxx, 2018-08-09:
Sounds about right.  If everything works for your needs, consider yourself extremely lucky!

As I think about it, the mouse stick might be configurable as a mouse device under your Linux X/Xorg configuration, as that mouse device typically acts as a replacement of using your usual typical mouse device within a game.  Never thought much of it now, as I haven't used the joystick under Linux/X since playing bf1942/wine decades ago.  The next latest and greatest is something call "ARMA 3" by Bohemia.  But their Linux game version is an extremely slimmed down version of the Windows' version, with joystick usage being apparently unconfigurable/non-detectable via the required Steam.  (Or a royal PITA, making it nearly impossible to use.)  Last I checked 6 months ago anyways.  If this ARMA 3 game ever gets fully working, almost guarenteed to have much more activity with (this) Saitek device support in Linux.