nirenjan/libx52 #17: how to fix: error while loading shared libraries: libx52.so.2

Xeraster, 2019-12-03:
So I am running Ubuntu 18.04. I followed the installation instructions in the readme but whenever I try to execute anything such as x52test or x52cli, it gives me an error saying "x52test: error while loading shared libraries: libx52.so.2: cannot open shared object file: No such file or directory"

Is there a way to fix this? Where do I get "libx52.so.2" and where do I put it?

rogerxxxx, 2019-12-07:
You're missing libx52.so.2.

This library appears to be compiled and installed by x52pro sources after 
compile & install.  Or, the library was installed in the incorrect location.

>   So I am running Ubuntu 18.04. I followed the installation instructions in
>   the readme but whenever I try to execute anything such as x52test or
>   x52cli, it gives me an error saying "x52test: error while loading shared
>   libraries: libx52.so.2: cannot open shared object file: No such file or
>   directory"
>
>   Is there a way to fix this? Where do I get "libx52.so.2" and where do I
>   put it?
>
>   [2]unsubscribe.[3][IMG]
>
>References
>
>   Visible links
>   1. https://github.com/nirenjan/x52pro-linux/issues/17?email_source=notifications&email_token=AC4KUVTUMWPZHDETARSECELQW3WBPA5CNFSM4JVAMYF2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4H526BRA

nirenjan, 2020-03-31:
The default location for the library is `/usr/local/lib/`, and this folder is not typically within the default search path for the loader.

One way to work around this is to run `./configure --prefix=/usr`, followed by `make` and `sudo make install`, this will ensure that the libraries are placed in `/usr/lib` instead of `/usr/local/lib`.

An alternative method is to run the following command (assuming Bash shell)

```
$ export LD_LIBRARY_PATH=/usr/local/lib
$ ./x52test
```

nirenjan, 2020-05-27:
You might also consider installing the Debian package from the release - see https://github.com/nirenjan/x52pro-linux/releases/download/v0.2.0/x52pro-linux1_0.2.0-1_amd64.deb

closed 2020-12-07 by stale[bot]