nirenjan/libx52 #19: Error: Unknown error -4

KishCom, 2020-05-18:
This library is exactly what I was looking for - thank you so much for developing it!

I've compiled and installed things OK, `x52test` successfully runs the LED brightness and MFD brightness tests (which, interestingly, seems to only dim the LEDs on the throttle/left side) -- but then cuts out with:

```
Testing LED FIRE
LED FIRE - OFF

led_state(LIBX52_LED_FIRE, LIBX52_LED_STATE_OFF) failed with 4
Received Illegal instruction signal, quitting...
```

Trying to toggle LEDs on/off does not seem to work `x52cli led B amber` :

```
Error: Unknown error -4
```

Other relevant details:
- Ubuntu 20.04
- 06a3:075c Saitek PLC X52 Flight Controller
- Replacement ps/2 cable between throttle-side/stick-side (thought it might be pertinent to mention since I noticed the LEDs on the flight-stick/right side didn't dim with the test)
- Added a custom udev rule that allows write access to USB without need for sudo

nirenjan, 2020-05-18:
The library doesn't support setting the individual LEDs on the X52, only on the X52 Pro. But you did find a bug in the x52cli and x52test apps.

As for the stick side not dimming along with the throttle, it seems to point to the replacement cable, but I don't know anything about the protocol or connectivity between the two to give a firm answer.

I'll push some changes shortly to fix the test order and error messages. Would you be willing to help in determining if we can control the individual LEDs on the X52?

KishCom, 2020-05-19:
> Would you be willing to help in determining if we can control the individual LEDs on the X52?

You bet! My C skills are subpar -- but point me in the right direction and I'd be glad to help.

As a note: I don't think the non-pro X52s have different colours of LEDs. I've never seen the LEDs on my flight stick be anything other than blue. I got a little excited when I thought I could change their colour. [This is what my X52 looks like](https://i.imgur.com/fH0M7OW.jpg).

nirenjan, 2020-05-19:
Awesome. I haven't found anything concrete on whether one could actually control the individual LEDs. Everything seems to point to it not being supported, but it might still be worth a shot to try the raw commands.

Use the following commands, and let me know what you see.

```
x52cli raw 0x00b8 0x0100
x52cli raw 0x00b8 0x0101
x52cli raw 0x00b8 0x0200
x52cli raw 0x00b8 0x0201
```

Repeat increasing the second byte all the way to 0x14 (20 decimal). These are the LED commands for the X52 pro. Given that every other command works on the X52 (see #11), *assuming Saitek implemented this in their firmware*, these should toggle the individual LEDs.

[Here](https://github.com/nirenjan/x52pro-linux/blob/f4a81aba83730d17e29cf051eb88b955b28bff40/docs/specs/vendor_api.md#leds) is the vendor documentation for the X52 Pro LEDs, if you want to refer to it.

KishCom, 2020-05-19:
Threw all those commands at it, and nothing happened. No error, no LED change, nothing.

Did some more digging and while I wasn't able to find concrete documented proof, a comparison of the Windows driver screen for the two devices appears to confirm what we suspect: [Windows Driver Screen for the X52](https://i.imgur.com/8czBPPZ.png) vs [Windows Driver Screen for the X52 Pro](https://i.imgur.com/pnp4VAB.jpg).

Toggle-able/addressable LEDs appears to be an undocumented feature for the X52-Pro. Thanks for taking the time to look into this!

closed 2020-05-19 by KishCom