nirenjan/libx52 #21: X52 Feature Report

nirenjan, 2020-06-11:
I'm looking to get the USB interface descriptor for the Saitek X52 (VID 06a3, PID 075c or PID 0255).

On Ubuntu, you can get this info by running `usbhid-dump -d 06a3`, which will dump the HID report descriptor for all the Saitek devices that are connected.

Pinging the users I know have X52's - @winiciuscota, @KishCom

KishCom, 2020-06-18:
As requested:

```
001:003:000:DESCRIPTOR         1592493925.728025
 05 01 09 04 A1 01 09 01 A1 00 09 30 09 31 15 00
 26 FF 07 75 0B 95 02 81 02 09 35 15 00 26 FF 03
 75 0A 95 01 81 02 09 32 09 33 09 34 09 36 15 00
 26 FF 00 75 08 95 04 81 02 05 09 19 01 29 22 15
 00 25 01 95 22 75 01 81 02 75 02 95 01 81 01 05
 01 09 39 15 01 25 08 35 00 46 3B 01 66 14 00 75
 04 95 01 81 42 05 05 09 24 09 26 15 00 25 0F 75
 04 95 02 81 02 C0 C0
```

nirenjan, 2020-06-18:
@KishCom, thank you so much for the report. It lines up with what I guessed [here](https://github.com/nirenjan/x52pro-linux/wiki/X52-Controller-Map).

I still need to figure out the button mapping, and that doesn't show up on the report. I'll see if I can tweak the readjoy program in the tools directory to give more info.

macdaddyo, 2020-07-04:
> I'm looking to get the USB interface descriptor for the Saitek X52 (VID 06a3, PID 075c or PID 0255).
>
> On Ubuntu, you can get this info by running `usbhid-dump -d 06a3`, which will dump the HID [...]

Hope this helps:

usbhid-dump -d 06a3                                                                                              
```
001:003:000:DESCRIPTOR         1593859249.976733
 05 01 09 04 A1 01 09 01 A1 00 09 30 09 31 15 00
 26 FF 03 75 0A 95 02 81 02 75 02 95 01 81 01 09
 35 15 00 26 FF 03 75 0A 95 01 81 02 09 32 09 33
 09 34 09 36 15 00 26 FF 00 75 08 95 04 81 02 05
 09 19 01 29 27 15 00 25 01 95 27 75 01 81 02 75
 05 95 01 81 01 05 01 09 39 15 01 25 08 35 00 46
 3B 01 66 14 00 75 04 95 01 81 42 05 05 09 24 09
 26 15 00 25 0F 75 04 95 02 81 02 C0 C0
```

nirenjan, 2020-07-04:
@macdaddyo, can you share the PID of your device? You can get that by running `lsusb -d 06a3: -v`. Thanks

macdaddyo, 2020-07-04:
`
Bus 001 Device 003: ID 06a3:0762 Saitek PLC Saitek X52 Pro Flight Control System
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x06a3 Saitek PLC
  idProduct          0x0762 Saitek X52 Pro Flight Control System
  bcdDevice            1.23
  iManufacturer           1 Saitek
  iProduct                2 Saitek X52 Pro Flight Control System
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              230mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     125
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
`

nirenjan, 2020-07-04:
@macdaddyo, no problem. Thanks for taking the time.

nirenjan, 2020-07-06:
Thanks to everybody who responded. I have enough information to proceed for now.

nirenjan, 2020-07-17:
@KishCom, @macdaddyo 

Can you build the latest sources and verify that `utils/evtest/x52evtest` generates the right output for your joystick? You will need to install `libhidapi-dev` on Ubuntu and `hidapi` on OSX if you don't already have it.

KishCom, 2020-07-17:
@nirenjan Looks good to me! Compiles and output working as expected on Ubuntu 20.04.

macdaddyo, 2020-07-19:
@nirenjan switched to master branch and everything works

closed 2020-08-01 by nirenjan