nirenjan/libx52 #25: Allow to specify the group used in udev rules

lbeltrame, 2020-12-25:
**Is your feature request related to a problem? Please describe.**
Not all Linux distributions use `plugdev` for groups allowed to access input devices. openSUSE, for example, has an `input` group.

**Describe the solution you'd like**
Keep `plugdev` as default, but allow changing this in a `configure` call.

**Describe alternatives you've considered**
Replacing the group manually before installing, but for any distribution doing packaging that would be probably extra effort, and potentially error prone.

nirenjan, 2020-12-25:
@lbeltrame, can you try the `feature/custom-udev-input-group` branch and see if it works as you intended? If this is OK, I'll merge it to the master branch.

Thanks for the report.

nirenjan, 2020-12-26:
I forgot to mention, I've updated the INSTALL.md file in that branch with the instructions for customizing the input group.

lbeltrame, 2020-12-26:
The file is correctly generated, as far as I can see. Thanks for the prompt adjustment!

nirenjan, 2020-12-26:
Fix merged in master branch.

closed 2020-12-26 by nirenjan