nirenjan/libx52 #27: hotas for dumb

Daebis, 2021-01-27:
🇬🇧 Hello here,
I just followed your installation procedure.

`sudo apt install automake autoconf gettext autopoint libhidapi-dev libtool libusb-1.0-0-dev pkg-config python3 git libx52pro0`

as well as antimicroX via the deb package 

in antimicroX all buttons and axis are recognized?

but I expected a graphical interface (even simplistic) to manage the leds and text through your program.

can you give me the instructions that I miss to arrive at a similar result as the official program saitek

Thanks in advance

**===============================================================================**

🇫🇷 Bonjour ici, 
je viens de suivre votre procédure d'installation.
`sudo apt install automake autoconf gettext autopoint libhidapi-dev libtool libusb-1.0-0-dev pkg-config python3 git libx52pro0`
ainsi que antimicroX via le paquet deb

dans anti micro l'ensemble des bouton et axe sont bien reconnu?
mais je m'attendais a une interface graphique (même simpliste) pour géré les led et texte via votre programme. 

pourrez vous me donnez les instruction qu'il me manque pour arrivé a un résultat similaire que le programme officiel saitek

D'avance merci

nirenjan, 2021-01-27:
I don't have expertise programming a graphical interface, but if you follow the [Build and install instructions](https://github.com/nirenjan/x52pro-linux/blob/master/INSTALL.md), you will have the necessary programs installed to control the LEDs and MFD text and clock via the command line.

You can see the [manual page for x52cli](https://nirenjan.github.io/x52pro-linux/x52cli.html) for more details

nirenjan, 2021-01-27:
If you want a graphical application, [this repository on Gitlab](https://gitlab.com/999eagle/gx52) is a good place to start.

closed 2021-02-08 by nirenjan