nirenjan/libx52 #28: MFD time doesn't advance

j-silver, 2021-02-07:
Hello, I'm not sure if this is a bug, or if I missed the instructions for setting the time or if the controller has no internal clock and needs to be constantly updated (e.g. by a background process).

When I set time and date with x52cli, for example at 10:00am, the clock remains at 10:00 forever.

Am I doing anything wrong?

nirenjan, 2021-02-07:
There is no internal clock in the X52, so a process has to keep updating the time every minute.

Let me know if the

j-silver, 2021-02-08:
Hi, no there is no unequivocal indication that it has an internal clock, although there's a section called **To change your clock settings** at page 11 of the Setup Guide, which is available [here](https://support.logi.com/hc/en-us/articles/360024317693--Getting-Started-X52-Professional-Space-Flight-H-O-T-A-S-).

nirenjan, 2021-02-08:
Seems like the Windows driver has a background application that updates the clock periodically. It's unlikely that this would be mentioned in the user documents. I only figured it out by tracing the USB protocol packets.

This is something I have in my backlog for this project - I just haven't had time to get to it :)

j-silver, 2021-02-08:
So what they call 'clock' it's just a chunk of memory which must be updated by the OS every minute... Anyway, I don't really miss this feature. :-)

Instead It would be nice if Elite Dangerous used it to provide useful information about the game, since the MFD can display a lot of characters and symbols. 
But even in that case, I don't think it would be very handy to have to look at MFD while playing.

nirenjan, 2021-02-11:
I've not played Elite Dangerous, but a quick google search seems to indicate a number of projects that link the X52 MFD with the game. That said, what I did see was that these projects were targeting Windows (or so I assume, because they are using .sln files).

Anyway, I'll go ahead and close this issue, since it's not a bug.

closed 2021-02-11 by nirenjan