nirenjan/libx52 #31: Debian Bullseye compile error

LittleCannon, 2021-08-05:
Hi,
I'm having trouble compiling X52pro.

I'm receiving this error after "make":

`In file included from libx52/x52_common.h:14,
                 from libx52/x52_control.c:18:
/usr/include/libusb-1.0/libusb.h:934:1: error: packed attribute is unnecessary for 'libusb_control_setup' [-Werror=packed]
  934 | } LIBUSB_PACKED;
      | ^
cc1: all warnings being treated as errors
make[2]: *** [Makefile:1650: libx52/la-x52_control.lo] Error 1
make[2]: Leaving directory '/home/xxx/Downloads/x52pro-linux-master'
make[1]: *** [Makefile:2390: all-recursive] Error 1
make[1]: Leaving directory '/home/xxx/Downloads/x52pro-linux-master'
make: *** [Makefile:1147: all] Error 2`

If you have some time, can you assist me?
Ideally, if you can make new deb file with daemon support that would be great.

nirenjan, 2021-08-05:
Hi @LittleCannon,

Thanks for your bug report. Can you try to rerun the build with the following flag to `configure` - `--enable-compile-warnings=yes`. Please also provide the version of the compiler that you are using.

Thanks

LittleCannon, 2021-08-05:
Hi,
Tnx for quick reply.
After rerun with suggested flag, everything went well. 

gcc (Debian 10.2.1-6) 10.2.1 20210110

LittleCannon, 2021-08-05:
Done.
All went well. No errors, just 2 warnings about doxyfile.

`warning: Tag 'TCL_SUBST' at line 270 of file 'Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 1121 of file 'Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u`

nirenjan, 2021-08-05:
Thank you, the Doxygen warnings can be safely ignored. I do need them for now, so that it can build on older versions of Doxygen.

I've got a few more changes planned before I can create a new release, but I have integrated systemd support into the latest commit. Please run `configure --prefix=/usr`, `make` and `sudo make install`.

Then run `sudo systemctl daemon-reload`, `sudo systemctl enable x52d`, `sudo systemctl start x52d`. This will ensure that the daemon service starts automatically on a reboot.

closed 2021-08-05 by nirenjan

LittleCannon, 2021-08-05:
It works perfectly.
Thank you very much!