nirenjan/libx52 #32: Got compilation error after make (ini.h missing)

gabolander, 2021-08-11:
Hi there, when trying to compile, I get this error:

```
gabo@skylake ~/github/x52pro-linux $ make
make  all-recursive
.......
gcc -DHAVE_CONFIG_H -I.    -I . -I ./libx52io -I ./libx52 -I ./libx52util -I ./lib/pinelog -DSYSCONFDIR=\"/usr/etc\" -DLOCALEDIR=\"/usr/share/locale\" -DLOGDIR=\"/usr/var/log\" -DRUNDIR=\"/usr/var/run\" -pthread -fno-strict-aliasing -Wnested-externs -Wmissing-prototypes -Wstrict-prototypes -Wdeclaration-after-statement -Wimplicit-function-declaration -Wold-style-definition -Wjump-misses-init -Wall -Wextra -Wundef -Wwrite-strings -Wpointer-arith -Wmissing-declarations -Wredundant-decls -Wno-unused-parameter -Wno-missing-field-initializers -Wformat=2 -Wcast-align -Wformat-nonliteral -Wformat-security -Wsign-compare -Wstrict-aliasing -Wshadow -Winline -Wmissing-format-attribute -Wmissing-noreturn -Winit-self -Wmissing-include-dirs -Wunused-but-set-variable -Warray-bounds -Wreturn-type -Wswitch-enum -Wswitch-default -Wduplicated-cond -Wduplicated-branches -Wlogical-op -Wrestrict -Wnull-dereference -Wdouble-promotion -Werror -Wno-suggest-attribute=format -Wno-error=unused-parameter -Wno-error=missing-field-initializers -g -O2 -MT daemon/x52d-x52d_config_parser.o -MD -MP -MF daemon/.deps/x52d-x52d_config_parser.Tpo -c -o daemon/x52d-x52d_config_parser.o `test -f 'daemon/x52d_config_parser.c' || echo './'`daemon/x52d_config_parser.c
daemon/x52d_config_parser.c:17:10: fatal error: ini.h: File o directory non esistente
   17 | #include "ini.h"
      |          ^~~~~~~
compilation terminated.
make[2]: *** [Makefile:2082: daemon/x52d-x52d_config_parser.o] Errore 1
make[2]: uscita dalla directory «/home/gabo/github/x52pro-linux»
make[1]: *** [Makefile:2485: all-recursive] Errore 1
make[1]: uscita dalla directory «/home/gabo/github/x52pro-linux»

```
It's probably my fault, for instance I may have some missing *-dev package, but _./configure --prefix=/usr_ gave me no errors, and I don't know what packages ini.h header come from... 
Suggestions? Thanks in advance.

Gabriele

nirenjan, 2021-08-11:
That seems like a bug in configure. Can you share your config.log file so I can see what the issue is?

The ini.h file should come from libinih-dev, assuming a Debian derived distribution. Please see INSTALL.md.

nirenjan, 2021-08-11:
FWIW, I did try to build on an older version of Ubuntu, and unfortunately, inih does not build with 16.04, seems like it needs a newer version of meson. If you have Ubuntu 18.04 or newer, you should be able to install `libinih-dev`

nirenjan, 2021-08-12:
I've pushed a commit that should cause configure to bail out earlier. If you don't see any more problems after installing libinih-dev, then please close the issue.

commit ff105250 (2021-08-12): Make inih a mandatory package.

AC_CHECK_HEADERS and AC_SEARCH_LIBS do not abort when the header or
library respectively are not found. Since inih is a hard requirement,
make sure that the action-if-not-found fields call AC_MSG_ERROR.

Fixes #32
closed 2021-08-12 by nirenjan