nirenjan/libx52 #33: 0.2.2. --- erratic behavior

LittleCannon, 2021-09-13:
**Describe the bug**
After daemon is started and X52Pro connected, on display, mode status (1,2, 3) starts rapidly blinking. Also in game X52Pro is erratic ( it feels like it is repeating inputs, rather then holding constant value).

**To Reproduce**
Steps to reproduce the behavior:
1. Just start x52d 0.2.2
2. take a look at LCD display
3. Blinking mode  status number

**Expected behavior**
Normal behavior 

**Screenshots**
If applicable, add screenshots to
[x52evtest-output.txt](https://github.com/nirenjan/x52pro-linux/files/7155226/x52evtest-output.txt)
 help explain your problem.

**Environment (please complete the following information):**
 - OS: Debian Sid
 - libusb version: libusb-1.0
 - X52 type: Bus 001 Device 004: ID 06a3:0762 Saitek PLC Saitek X52 Pro Flight Control System

**Additional context**
In attachment x52evtest output,without touching hotas. It starts soon as hotas is connected.

nirenjan, 2021-09-13:
As of version 0.2.2, the daemon doesn't do anything but control the brightness, clock and LED states. The mode status number is also controlled in hardware, not by the daemon. I'm not sure why you are seeing the problem only when running the daemon.

Can you also attach the x52d logs? If you are using the released .deb package (or if you've built from source with the default configuration), then you'll need to edit `/lib/systemd/system/x52d.service`, and update the `ExecStart` line from `-v` to `-vvvv`. Then you can capture the logs by running `sudo journalctl -b -u x52d.service`.

If you're running the service manually, then you can append `-vvvv -l /tmp/x52d.log`, and attach that file.

nirenjan, 2021-09-13:
Also, without starting the daemon, can you capture `x52evtest` output? Also, please run `x52test` - this is a program which will exercise the hardware and make sure that the device is working as expected. The output that you attached is showing flapping on the MODE_3 button, which leads me to think that either the button sensor is not working as expected, or (more likely), the PS/2 cable connecting the throttle and stick is bad. I had to replace the cable on my own joystick several years ago, and I suspect that this might be the more likely reason you are seeing issues.

LittleCannon, 2021-09-13:
Tnx for really quick reply

My first thought was also that switch or even cable might be faulty, but as you can see from evtest with daemon stopped, it is not.

Please let me know if I can do anything more to assist.

[journactl_LOG.txt](https://github.com/nirenjan/x52pro-linux/files/7156140/journactl_LOG.txt)
[x52evtest_daemon_stop.txt](https://github.com/nirenjan/x52pro-linux/files/7156145/x52evtest_daemon_stop.txt)

LittleCannon, 2021-09-13:
oh, I forgot, I've built it from source.

nirenjan, 2021-09-13:
OK, unfortunately, the journalctl logs are getting cut off, can you please capture the output using `sudo journalctl -b -u x52d.service > journalctl_log.txt` and attach that output please?

LittleCannon, 2021-09-13:
[journalctl_log.txt](https://github.com/nirenjan/x52pro-linux/files/7156507/journalctl_log.txt)

nirenjan, 2021-09-13:
Huh... the journal logs are showing an error which was fixed before I released 0.2.2, what commit are you at?

LittleCannon, 2021-09-13:
I downloaded source from latest release. 0.2.2. I guess?

Also I tried with deb package as well. Same thing.

nirenjan, 2021-09-14:
I think I may have found the issue. Can you try with the commit `0a30bb1` on branch `check-device-attached` and see if you still see the problem?

LittleCannon, 2021-09-14:
You got it!
It works as expected now.

Thank you very very much!

nirenjan, 2021-09-14:
Thank you for the confirmation. If you can provide the verbose daemon logs when unplugging and reinserting the X52, I'd appreciate that very much.

LittleCannon, 2021-09-14:
Of course!

Here it is.
[journalctl_log.txt](https://github.com/nirenjan/x52pro-linux/files/7163483/journalctl_log.txt)

nirenjan, 2021-09-14:
Thank you. One last request - can you share the contents of `libx52/x52_stringify.c`? The log messages that you shared indicate another fault which I thought I fixed in that file.

closed 2021-09-14 by nirenjan

LittleCannon, 2021-09-14:
Yes no problem.
*********
/*
 * Saitek X52 Pro MFD & LED driver - stringification
 *
 * Copyright (C) 2012-2021 Nirenjan Krishnan (nirenjan@nirenjan.org)
 *
 * SPDX-License-Identifier: GPL-2.0-only WITH Classpath-exception-2.0
 */

#include "config.h"
#include <stdio.h>

#include "libx52.h"
#include "gettext.h"

#define N_(str) gettext_noop(str)
#define _(str)  dgettext(PACKAGE, str)

#define STRINGIFY(name, max_id, errstr, ...) \
const char * libx52_ ## name ## _to_str (libx52_ ## name param) { \
    static char invalid[256]; \
    static const char *desc[] = { __VA_ARGS__ }; \
    if (param >= 0 && param <= max_id) { \
        return _(desc[param]); \
    } \
    snprintf(invalid, sizeof(invalid), _(errstr), param); \
    return invalid; \
}

STRINGIFY(clock_id, LIBX52_CLOCK_3, N_("Unknown clock ID %d"),
    N_("primary"),
    N_("secondary"),
    N_("tertiary"),
)

STRINGIFY(clock_format, LIBX52_CLOCK_FORMAT_24HR, N_("Unknown clock format %d"),
    N_("12 hour"),
    N_("24 hour"),
)

STRINGIFY(date_format, LIBX52_DATE_FORMAT_YYMMDD, N_("Unknown date format %d"),
    N_("DD-MM-YY"),
    N_("MM-DD-YY"),
    N_("YY-MM-DD"),
)

STRINGIFY(led_state, LIBX52_LED_STATE_GREEN, N_("Unknown LED state %d"),
    N_("off"),
    N_("on"),
    N_("red"),
    N_("amber"),
    N_("green"),
)

const char * libx52_led_id_to_str(libx52_led_id id)
{
    static char invalid[256];

    switch (id) {
    case LIBX52_LED_FIRE:
        return _("Fire");

    case LIBX52_LED_A:
        return _("A");

    case LIBX52_LED_B:
        return _("B");

    case LIBX52_LED_D:
        return _("D");

    case LIBX52_LED_E:
        return _("E");

    case LIBX52_LED_T1:
        return _("T1");

    case LIBX52_LED_T2:
        return _("T2");

    case LIBX52_LED_T3:
        return _("T3");

    case LIBX52_LED_POV:
        return _("POV");

    case LIBX52_LED_CLUTCH:
        return _("Clutch");

    case LIBX52_LED_THROTTLE:
        return _("Throttle");

    default:
        snprintf(invalid, sizeof(invalid), _("Unknown LED ID %d"), id);
        return invalid;
    }
}

LittleCannon, 2021-09-14:
Oh btw, if I can assist more, please let me know.
Really it is no trouble for me.

nirenjan, 2021-09-14:
Thank you, if I need more help, I'll raise a separate discussion and tag you.

commit 7d757dd4 (2021-09-14): Disable device check routine in daemon

Prior to this change, the device check was sending a vendor specific
command with wIndex and wValue both set to 0 every 50 ms. On some
systems, this was causing issues with the joystick flapping the state,
and reporting weird values from the stick, and generally sluggish
response.

This change uses the updated libx52 library which uses the hotplug
notification to determine if the device is connected, and should resolve
the issues seen.

Fixes #33