nirenjan/libx52 #35: ./configure --prefix=/usr tries to install logs in /usr/var/log, fails installation ("missing operand")

lbeltrame, 2021-09-15:
**Describe the bug**
When configuring with `--prefix=/usr`, the Makefile is set up in a way that will try to create the log directory in `/usr/var/log` rather than `/var/log`. It will also fail (see output).

**To Reproduce**
Steps to reproduce the behavior:
1. `./configure --prefix=/usr`
2. `make`
3. `sudo make install`

**Expected behavior**
Installation should finish and not try to create bogus directories.

**Environment (please complete the following information):**
 - openSUSE Tumbleweed

**Additional context**

```
# sudo make install
[...]
 /usr/bin/mkdir -p '/usr/bin'
  /bin/sh ./libtool   --mode=install /usr/bin/install -c x52cli x52test x52evtest x52d '/usr/bin'
libtool: install: /usr/bin/install -c .libs/x52cli /usr/bin/x52cli
libtool: install: /usr/bin/install -c .libs/x52test /usr/bin/x52test
libtool: install: /usr/bin/install -c .libs/x52evtest /usr/bin/x52evtest
libtool: install: /usr/bin/install -c .libs/x52d /usr/bin/x52d
make  install-exec-hook
make[3]: Entering directory '/home/lb/Download/Sources/x52pro-linux'
/usr/bin/mkdir -p /usr/var/log
/usr/bin/mkdir -p 
/usr/bin/mkdir: missing operand
Try '/usr/bin/mkdir --help' for more information.
```

lbeltrame, 2021-09-15:
The issue in Makefile (unfortunately I can't speak autotools):

```
install-exec-hook:
	$(MKDIR_P) $(DESTDIR)$(localstatedir)/log
	$(MKDIR_P) $(DESTDIR)$(runstatedir)
```

`localstatedir` is seen as a subdirectory of `DESTDIR` (but may not be), and `runstatedir` is undefined.

lbeltrame, 2021-09-15:
That's because in some autotools version, `runstatedir` is not defined, see https://github.com/troglobit/sysklogd/issues/6 for context, for example.

nirenjan, 2021-09-15:
Thank you for the bug report. I've pushed a fix to the `mouse-support` branch, and should be able to get it into `master` once that branch is tested.

nirenjan, 2021-09-15:
For what it's worth, autoconf defaults `localstatedir` to `$(prefix)/var`. I've also updated the build instructions to reflect that.

lbeltrame, 2021-09-17:
I confirm that the issue is no longer present in `mouse-support` as I've been obviously been able to test that.

nirenjan, 2021-09-19:
Fix merged to `master` branch.

closed 2021-09-19 by nirenjan