nirenjan/libx52 #36: X52Pro Second POV hat (metal one) LED

LittleCannon, 2021-09-15:
**Is your feature request related to a problem? Please describe.**
Unable to change LED color on second POV hat (metal one)

**Describe the solution you'd like**
To be able to change LED color

**Describe alternatives you've considered**
---

**Additional context**
---

nirenjan, 2021-09-15:
I don't believe the second POV LED can be controlled. There is only a blink feature, which blinks it on and off, but I haven't found any method to actually change the color. This is pretty much based on the Windows control panel interface, which can change the color of every LED (or On/Off only for the Fire button & throttle LEDs), and that doesn't have a control for the second POV.

That said, it might need a little reverse engineering to see if there is a hidden control message that can control that LED. I just haven't tried it.

nirenjan, 2021-09-15:
I gave it a shot, but it seems like a limitation of the hardware that you cannot control the hat LED.

nirenjan, 2021-09-15:
Seems like a hardware limitation, I haven't found any control message that can modify the hat LED, apart from the blink.

LittleCannon, 2021-09-15:
Understood,
I was just playing with new configurations and profiles, and I noticed that HAT LED is not on the list.

Tnx for trying.

closed 2021-09-15 by LittleCannon