nirenjan/libx52 #38: Default log is too verbose

lbeltrame, 2021-10-01:
**Describe the bug**
When I don't have my X52 connected, the journalctl logs are spammed with

```
ott 01 22:57:58 leon.lan x52d[910]: TRACE: x52d_device.c:37 Attempting to connect to X52 device
ott 01 22:57:58 leon.lan x52d[910]: TRACE: x52d_device.c:44 No compatible X52 device found
ott 01 22:57:58 leon.lan x52d[910]: TRACE: x52d_device.c:46 Sleeping for 5 seconds before trying to acquire device again
ott 01 22:57:58 leon.lan x52d[910]: TRACE: x52d_io.c:58 Device disconnected, trying to connect
ott 01 22:57:58 leon.lan x52d[910]: TRACE: x52d_io.c:65 No compatible X52 I/O device found. Sleeping 5 seconds before trying again.
```
I haven't found a way to disable this logging.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the daemon without a joystick attached

**Expected behavior**
No TRACE logs unless explicitly enabled.

nirenjan, 2021-10-02:
I thought I had disabled that. Please edit
/lib/systemd/system/x52d.service, and remove the -vvv from the ExecStart
line.

nirenjan, 2021-10-06:
I've pushed a fix to the master branch. Please pull from the tip of master.

closed 2021-10-06 by nirenjan

lbeltrame, 2021-10-06:
Whoops, sorry. I had actually tested your suggestion but forgot to get back to you (and of course, it worked). Thanks!