nirenjan/libx52 #40: Mouse mini-stick as additional axes?

LittleCannon, 2022-02-16:
Hi,
One quick question.

Is it possible to define mini-stick as additional axes?
I'm playing Elite Dangerous and it would be great to be able to define mini-stick as analogue lateral thrust control.

nirenjan, 2022-02-16:
That's an interesting thought. Can you give the kernel module a try for now? It maps the mouse stick to ABS_TILT_X and ABS_TILT_Y, so it may get detected as an additional axis.

LittleCannon, 2022-02-16:
I've quickly recompiled fresh version and, according to evtest, it is still  detected ABS_THUMBY and ABS_THUMBX

Also, I tried in game, with and without mouse option enabled, it is not registering any movement as axis.

nirenjan, 2022-02-16:
Hmm, can you confirm the output of `lsmod` and `uname -a`? If you build the kernel module against your kernel (just cd to the `kernel_module` directory and run `make`), and insmod it, it should change the axis reported by the kernel driver.

The default kernel driver will report the mouse stick as ABS_MISC and ABS_MISC+1, which ED may not support.

nirenjan, 2022-02-16:
Also, it appears that you are using `x52evtest`, that is just decoding the raw report from the joystick - it doesn't change anything from the kernel, while ED is probably reading it from the kernel instead. You want to use `evtest`, this is a system utility that will read the events from the kernel.

LittleCannon, 2022-02-16:
I'm on 5.16 kernel (Debian Sid)
5.16.0-1-amd64 #1 SMP PREEMPT Debian 5.16.7-2 (2022-02-09) x86_64 GNU/Linux

OK, partial success.
After insmoding it, I was able to assign X axis, but it is recognized in game as slider above mouse stick
[image]
[evtest.txt](https://github.com/nirenjan/libx52/files/8083040/evtest.txt)

LittleCannon, 2022-02-16:
Oh, and if it will be of any use, here is evtest before insmod.
As you can see, Y axis is (not)recognized as "?"
[evtes-before_insmodt.txt](https://github.com/nirenjan/libx52/files/8083117/evtes-before_insmodt.txt)

nirenjan, 2022-02-16:
Hmm, I don't know how ED is mapping the axes, I don't play it myself. It seems like the kernel module is doing the right thing - but ED is not recognizing the axes. Another thing I've found is that if you are launching ED through Steam, then Steam may be mapping the controller in a different way, so you may want to try disabling Steam's controller logic.

LittleCannon, 2022-02-17:
I did disable Steam Controllers, but it appears it is ED thing.
Never mind, it was just an idea.
I'll close this issue now.

closed 2022-02-17 by LittleCannon

nirenjan, 2022-02-17:
Feel free to play around with the reporting in the kernel module. It may just be that ED recognizes a different set of axes than what the kernel module reports. Here are the known axes from the kernel:

```
ABS_X
ABS_Y
ABS_Z
ABS_RX
ABS_RY
ABS_RZ
ABS_THROTTLE
ABS_RUDDER
ABS_WHEEL
ABS_GAS
ABS_BRAKE
ABS_HAT0X
ABS_HAT0Y
ABS_HAT1X
ABS_HAT1Y
ABS_HAT2X
ABS_HAT2Y
ABS_HAT3X
ABS_HAT3Y
ABS_PRESSURE
ABS_DISTANCE
ABS_TILT_X
ABS_TILT_Y
ABS_TOOL_WIDTH
ABS_VOLUME
ABS_MISC
```

It's likely that `ABS_PRESSURE`, `ABS_DISTANCE`, `ABS_TOOL_WIDTH` and `ABS_VOLUME` may not be helpful, but the others might be a good set to try.