nirenjan/libx52 #41: build failed with doxygen enabled: missing docs/Makefile.am

maxb-odessa, 2022-05-02:
make[2]: Entering directory '/home/maxb/src/x52pro-linux'
doxygen Doxyfile
warning: Tag 'TCL_SUBST' at line 270 of file 'Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 1123 of file 'Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Cannot open file tab_a.png for writing
Warning: Cannot open file tab_b.png for writing
Warning: Cannot open file tab_h.png for writing
Warning: Cannot open file tab_s.png for writing
Warning: Cannot open file nav_h.png for writing
Warning: Cannot open file nav_f.png for writing
Warning: Cannot open file bc_s.png for writing
Warning: Cannot open file closed.png for writing
Warning: Cannot open file open.png for writing
Warning: Cannot open file bdwn.png for writing
Warning: Cannot open file sync_on.png for writing
Warning: Cannot open file sync_off.png for writing
error: Could not open file /home/maxb/src/x52pro-linux/docs/html/doxygen.css for writing
make[2]: *** [Makefile:3654: docs/.stamp] Error 1
make[2]: Leaving directory '/home/maxb/src/x52pro-linux'
make[1]: *** [Makefile:2805: all-recursive] Error 1
make[1]: Leaving directory '/home/maxb/src/x52pro-linux'
make: *** [Makefile:1286: all] Error 2

nirenjan, 2022-05-02:
What OS are you using? Versions of packages?

I don't see this issue in the CI build, which is building on both Ubuntu 18.04 and 20.04

nirenjan, 2022-05-02:
I tested in an Arch Linux and Ubuntu 22.04 container as well, no issues seen.

maxb-odessa, 2022-05-02:
Just cloned the repo into new dir and the error is gone.

Old dir branches:
maxb@flatbox$ git branch -a
  gh-pages
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/debian-packaging
  remotes/origin/driver
  remotes/origin/feature/alt_lookup_engine
  remotes/origin/feature/cpp-library
  remotes/origin/feature/daemon
  remotes/origin/feature/non-recursive-automake
  remotes/origin/feature/test-new-mem-api
  remotes/origin/gh-pages
  remotes/origin/gh-pages-autogen
  remotes/origin/logging-changes
  remotes/origin/master
  remotes/origin/notify-socket
  remotes/origin/x52dcomm
  remotes/origin/x65f-driver

Newely cloned dir branches:
maxb@flatbox$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/debian-packaging
  remotes/origin/feature/alt_lookup_engine
  remotes/origin/feature/cpp-library
  remotes/origin/feature/test-new-mem-api
  remotes/origin/gh-pages
  remotes/origin/logging-changes
  remotes/origin/master
  remotes/origin/notify-socket

Sorry for the inconvenience

nirenjan, 2022-05-02:
No problem. Glad it's not something that's actually broken :)

closed 2022-05-02 by nirenjan