nirenjan/libx52 #44: x52d mouse diagonal cursor speed

VorpalBlade, 2022-06-05:
**Is your feature request related to a problem? Please describe.**
1. Run x52d with mouse support enabled and speed in config set to 15
2. Connect Saitek X52 Pro
3. Notice that horizontal and vertical movements are much slower than diagonal movements (45 degrees diagonal shows this most clearly).

**Describe the solution you'd like**

It feels like the speed is handled per axis. To me (given how the mouse joystick feels) I think it should be handled as "max total speed". As it is now it is very awkward to use.

It may be worth investigating how trackpoints on Thinkpads (and similar laptops) handle this, as those feel way more natural in their diagonal movements, and they are essentially also small mouse joysticks.

**Describe alternatives you've considered**

I considered implementing this myself and making a PR, but I want to discuss this first. I would also need to do research on how this works with track points (and if that is handled in software or hardware in that case).

**Additional context**

If it matters, my X52 Pro is really old, probably from the first or second year they were manufactured.

nirenjan, 2022-06-05:
You're right, the axis movements are handled independently. One thing we could check is if your hardware is working as expected.

When you run x52evtest, observe the values printed in the console, especially when you move the mouse stick in the different directions. Does the value for THUMBX and THUMBY match when you move the stick in the cardinal directions vs diagonally?

VorpalBlade, 2022-06-05:
Using `x52evtest` is really difficult. Something seems to be wrong with the joystick twist axis (which I don't use, I have rudder pedals) and it generates constant noise when centred (ABS_RZ events ranging from 470 to 545), though that axis is still when twisted significantly far enough away from the centre position . :-(

After filtering that with grep, both of the mouse axes seem to have values ranging from 0-15, though I have to press fairly hard to get to the outermost values (1-14 is more typical). The values seem to behave sensibly.

When I press really hard on the mouse thumbstick I can also feel that there is a "hard stop" that is square (i.e. as I slide it around I can feel the corners on the diagonal). This is in contrast to the trackpoint of my thinkpads (plural, I have used thinkpads for many years, and I don't throw away old computers that still work), where I can't feel such hard stops. After some investigation I realised that the actual trackpoints don't seem to move at all, only the rubber dome on top of it seems to be squish and stretch a bit as they are made of a flexible material. I believe the plastic nub under the rubber dome is actually fixed and uses some sort of pressure or force sensors.

The different construction methods does of course result in completely different feel, and it is likely that it affects how the resistance of the axis works as well. Thinking about the physics I believe that on a trackpoint the force I need to apply will be proportional to the deviation from the centre (Euclidian distance, L2 norm).

In contrast: on the X52 where the axes that actually move, the force I need to apply will be per axis (if it increases at all before hitting the hard stop, not sure about that). Thus the force is additive between the axes (Manhattan distance, L1 norm). This means that when it *feels like* I press with an equal amount of force in a cardinal direction and a diagonal direction, I'm actually moving further away from the centre position when going diagonal.

In conclusion, though I'm not sure yet, I now suspect this is just how the hardware is and perhaps nothing can be done about it. I may simply have to get used to it. Unless I want to make an electronics project and transplant a Thinkpad trackpoint onto the X52 ;-)

VorpalBlade, 2022-12-16:
As far as I know this issue persists. I haven't used my x52 since then (I have not had any time for flight sims in recent months). As such I don't remember, did you need anything further from me?

nirenjan, 2022-12-16:
I haven't had time to work on it either, the stale bot just kicked in automatically.