nirenjan/libx52 #46: x52d: a single scroll event may result in multiple REL_WHEEL reports

nirenjan, 2022-06-14:
**Observed behavior**
With some hardware, the joystick may report multiple events with the wheel button selected. As a result, x52d receives multiple events where the scroll button is on, but with different values in the axis reports. The following log reported by @VorpalBlade in #45 has some sample events: [scrolling.log](https://github.com/nirenjan/libx52/files/8847098/scrolling.log).

x52d currently only looks at the current state of the scroll buttons, therefore, faulty hardware could result in several virtual wheel events sent to the uinput node for a single scroll event on the joystick wheel.

**Expected behavior**
Each scroll event from the joystick should result in only 1 scroll event on the virtual mouse.

VorpalBlade, 2022-12-16:
As far as I know this issue persists. I haven't used my x52 since then (I have not had any time for flight sims in recent months). As such I don't remember, did you need anything further from me?

nirenjan, 2023-03-18:
@VorpalBlade , can you give the changes in the `fix-multiple-scroll-events` branch a try? I'm hoping that this should fix the issue you were seeing.

nirenjan, 2026-04-03:
No rush. I suspect this is now fixed with the changes I made with edge detection, but I would like confirmation with a device that's known to input spurious events.