nirenjan/libx52 #47: FATAL: x52d_main.c:117 Daemon is already running as PID 22828

MNS26, 2022-12-23:
whenever i run any commands it will throw a fatal message (`FATAL: x52d_main.c:117 Daemon is already running as PID 22828`)

when running anything other than `x52d -h` it throws a permission error

when using `sudo x52d` running any command after that (other than `x52d -h`) it will give the "FATAL" error
[image]

nirenjan, 2022-12-23:
Did you install the package from source or the Debian package?

Can you run `ps aux | grep x52d` and share the output. Also the output of `systemctl status x52d` please.

MNS26, 2022-12-23:
i installed it from the AUR on arch

MNS26, 2022-12-23:
the output of both 
[image]

MNS26, 2022-12-23:
i can try and install the github version if needed

nirenjan, 2022-12-23:
For some reason, it looks like the daemon might have crashed earlier and not cleaned up the PID file - look for a file `/var/run/x52d.pid` and if it exists, delete it.

nirenjan, 2022-12-23:
BTW, this is fixed in the latest version on Github.

MNS26, 2022-12-23:
there was a pid file yeah... 
i will install the git version them

MNS26, 2022-12-23:
[image]
hmm after install it thew errors and didnt actually install

MNS26, 2022-12-24:
[image]
welp installed the github version and its the same thing

MNS26, 2022-12-24:
[image]
when i killed x52d it shows one with my user but... throws the `fopen: permission denied` message

MNS26, 2022-12-24:
@nirenjan hmm how does it handle when no x52 is plugged in?

nirenjan, 2022-12-26:
I did fix #43 in the latest version on Github, so that may be the reason why the 0.2.3 version crashed. I haven't gotten around to making a new release yet though. One thing though, if you installed from Github, you should stick to running it via systemd, that will ensure it runs in foreground mode, rather than background, and it will skip the check for the PID file.

nirenjan, 2022-12-26:
The daemon is capable of recognizing when no joystick is connected, and it will try to reconnect every 5 seconds or so. If a joystick is connected, then it will update the joystick every 50 ms, and that should detect a disconnection nearly instantly. There was a bug in earlier versions which didn't quite handle the disconnect and reconnect properly, so that's why you should switch to the latest version.

nirenjan, 2022-12-26:
@MNS26, could you please uninstall the Github version and update to v0.3.0 from AUR? Make sure you cleanup the PID file.

MNS26, 2022-12-26:
> @MNS26, could you please uninstall the Github version and update to v0.3.0 from AUR? Make sure you cleanup the PID file.

how do i uninstall the git version?
`sudo make uninstall?`

MNS26, 2022-12-26:
> I did fix #43 in the latest version on Github, so that may be the reason why the 0.2.3 version crashed. I haven't gotten around to making a new release yet though. One thing though, if you installed [...]

i never had a issue with it (having it plugged in or not when starting the daemon didnt make a difference)

nirenjan, 2022-12-26:
> > @MNS26, could you please uninstall the Github version and update to v0.3.0 from AUR? Make sure you cleanup the PID file.
> 
> how do i uninstall the git version?
> `sudo make uninstall?`

Yes, `sudo make uninstall `

MNS26, 2022-12-26:
just did a uninstall and no pid files left

rn installing AUR version
[image]

MNS26, 2022-12-26:
also for some odd reason Elite Dangerous sees the controls double (still works fine)
[image]

MNS26, 2022-12-26:
oh that reminds me... is there _any_ way to disable the little mouse knob on the throttle?

accidentally bumping it makes the game go nuts (axis 8 & 9) and for some reason i cant seem to assign them in game 
(just a slight bump)
[image]

MNS26, 2022-12-26:
Just finished install
[image]

nirenjan, 2022-12-31:
You should already be running x52d from systemd, check `systemctl status x52d.service`. Also, x52d requires root permissions to write to certain directories.

MNS26, 2022-12-31:
hmm i was kinda hoping to not have to rely on root to access it

after manually starting the service (either via `sudo x52d` or with `systemctl`) it works

on another note... do you have a communication example for sending data _to_ the x52? (planning on replacing it with a RP2040 cus if its 12bit ADC)

MNS26, 2023-01-01:
hmm idk whats going on but the trigger on the joystick is set to half pressed and T2...
this happened after enabling the service
[image]
</r>
[image]
i also cant seem to bind axis 7 & 8 (not that i want to)

MNS26, 2023-01-01:
hmm yeah for whatever reason it now keeps sending wrong inputs

MNS26, 2023-01-01:
[image]

the button inputs from top to botton
mid Dpad up
mid Dpad down
mid Dpad left
mid Dpad right
C
B
A
left Dpad right
left Dpad left 

all of those come from the joystick
running `sdl2-jstest -t 0` shows all the correct inputs and buttons tho

its currently 11:45pm ill look into it more tomorrow

MNS26, 2023-01-02:
welp i think i found the issue... 
i have no clue how this happened but... sees the X52 as a raspberry pi pico instead 
(i did fool with the PID and VID of the pico but its not plugged in)
[image]

nirenjan, 2023-01-03:
I've not played Elite Dangerous, but if I understand correctly, it is running under Wine/Proton, and we need a way to clear its cache of joysticks. I'm sorry I couldn't be of more help here.

MNS26, 2023-01-03:
i just reinstalled the OS to be sure it wasnt that and its still there so... ill have a look on google

MNS26, 2023-01-03:
should i install the kernel module for the X52 pro?  (the one in the repo folder)
im running the latest zen kernel

nirenjan, 2023-01-03:
No, the kernel module is no longer needed. Glad you found the problem with the bindings

MNS26, 2023-01-03:
hmm weird... cus it sees the mouse X and Y axis as a joystick and making proton go ballistic 
but with it it doesnt but... itt indeed causes the buttons to be rearranged and thus a not so happy game
with the module
[image]
 
 without the module
 
[image]

those 2 extra axis cause nearly all the buttons to be in the pressed state to proton... 
is there a way to preserve the button layout but remove the 2 axis? (they are mapped to mouse with the `x52d.service`)

nirenjan, 2023-01-03:
You don't need the kernel module at all - it was an exercise in learning kernel module programming way back, I will remove it, since it's no longer needed in any recent kernel. `x52d` does not use the kernel provided events, it interprets the data directly from the raw USB report provided by `hidraw`.

For the mouse, the axis themselves are handled by the kernel, `x52d` just uses the events to trigger virtual mouse events, that is, the mouse can be controlled in Linux using the thumbstick. You can disable it by editing `/etc/x52d/x52d.conf` and reloading the daemon - `systemctl reload x52d.service`

MNS26, 2023-01-03:
the mouse indeed works with the `x52d.service` BUT... it also shows up as axis 7 and 8 and proton gets VERY confused about the button presses.
with the kernel module the mouse still works but axis 7 and 8 dont show up but do work as mouse... otherwise i actually cant use the mouse nub at all

MNS26, 2023-01-03:
without the kernel driver on the latest zen kernel
`Linux ROG-STRIX-SCAR-II 6.1.2-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Sat, 31 Dec 2022 17:40:33 +0000 x86_64 GNU/Linux`

nub pointing top right
[image]

nub pointing top left 
[image]

nub pointing bottom right
[image]

nub pointing bottom left
[image]

nirenjan, 2023-01-03:
Yeah, the kernel module was written a while back and it reorders the axis and button maps, so not ideal. I haven't used it in a while, so I'm not sure if there is a bug in the parsing that would cause multiple button presses to get triggered. Best not to use it at all.

MNS26, 2023-01-03:
this is WITHOUT the kernel module....
WITH the kernel module i dont have that issue

nirenjan, 2023-01-03:
Uh-oh, that doesn't sound right at all. Could you paste the output of `x52bugreport`? I'm assuming you have the latest release, i.e. 0.3.0

MNS26, 2023-01-03:
also regarding the button mapping, at least for the x52 pro

```
#define X52_TRIGGER_1       BTN_TRIGGER_HAPPY1      //BTN_TRIGGER_HAPPY1
#define X52_TRIGGER_2       BTN_TRIGGER_HAPPY15     //BTN_TRIGGER_HAPPY2
#define X52_BTN_FIRE        BTN_TRIGGER_HAPPY2      //BTN_TRIGGER_HAPPY3
#define X52_BTN_A           BTN_TRIGGER_HAPPY3      //BTN_TRIGGER_HAPPY4
#define X52_BTN_B           BTN_TRIGGER_HAPPY4      //BTN_TRIGGER_HAPPY5
#define X52_BTN_C           BTN_TRIGGER_HAPPY5      //BTN_TRIGGER_HAPPY6
#define X52_BTN_D           BTN_TRIGGER_HAPPY7      //BTN_TRIGGER_HAPPY7
#define X52_BTN_E           BTN_TRIGGER_HAPPY8      //BTN_TRIGGER_HAPPY8
#define X52_BTN_PINKIE      BTN_TRIGGER_HAPPY6      //BTN_TRIGGER_HAPPY9
#define X52_BTN_T1_UP       BTN_TRIGGER_HAPPY9      //BTN_TRIGGER_HAPPY10
#define X52_BTN_T1_DN       BTN_TRIGGER_HAPPY10     //BTN_TRIGGER_HAPPY11
#define X52_BTN_T2_UP       BTN_TRIGGER_HAPPY11     //BTN_TRIGGER_HAPPY12
#define X52_BTN_T2_DN       BTN_TRIGGER_HAPPY12     //BTN_TRIGGER_HAPPY13
#define X52_BTN_T3_UP       BTN_TRIGGER_HAPPY13     //BTN_TRIGGER_HAPPY14
#define X52_BTN_T3_DN       BTN_TRIGGER_HAPPY14     //BTN_TRIGGER_HAPPY15
#define X52_BTN_CLUTCH      BTN_TRIGGER_HAPPY31     //BTN_TRIGGER_HAPPY16
#define X52_MOUSE_LEFT      BTN_TRIGGER_HAPPY16     //BTN_TRIGGER_HAPPY17
#define X52_MOUSE_RIGHT     BTN_TRIGGER_HAPPY19     //BTN_TRIGGER_HAPPY18
#define X52_MOUSE_FORWARD   BTN_TRIGGER_HAPPY17     //BTN_TRIGGER_HAPPY19
#define X52_MOUSE_BACKWARD  BTN_TRIGGER_HAPPY18     //BTN_TRIGGER_HAPPY20
#define X52_STICK_POV_N     BTN_TRIGGER_HAPPY20     //BTN_TRIGGER_HAPPY21
#define X52_STICK_POV_E     BTN_TRIGGER_HAPPY21     //BTN_TRIGGER_HAPPY22
#define X52_STICK_POV_S     BTN_TRIGGER_HAPPY22     //BTN_TRIGGER_HAPPY23
#define X52_STICK_POV_W     BTN_TRIGGER_HAPPY23     //BTN_TRIGGER_HAPPY24
#define X52_THROT_POV_N     BTN_TRIGGER_HAPPY24     //BTN_TRIGGER_HAPPY25
#define X52_THROT_POV_E     BTN_TRIGGER_HAPPY25     //BTN_TRIGGER_HAPPY26
#define X52_THROT_POV_S     BTN_TRIGGER_HAPPY26     //BTN_TRIGGER_HAPPY27
#define X52_THROT_POV_W     BTN_TRIGGER_HAPPY27     //BTN_TRIGGER_HAPPY28
#define X52_MODE_1          BTN_TRIGGER_HAPPY28     //BTN_TRIGGER_HAPPY29
#define X52_MODE_2          BTN_TRIGGER_HAPPY29     //BTN_TRIGGER_HAPPY30
#define X52_MODE_3          BTN_TRIGGER_HAPPY30     //BTN_TRIGGER_HAPPY31
#define X52_BTN_FUNCTION    BTN_TRIGGER_HAPPY32     //BTN_TRIGGER_HAPPY32
#define X52_BTN_START_STOP  BTN_TRIGGER_HAPPY33     //BTN_TRIGGER_HAPPY33
#define X52_BTN_RESET       BTN_TRIGGER_HAPPY34     //BTN_TRIGGER_HAPPY34
#define X52_BTN_PG_UP       BTN_TRIGGER_HAPPY35     //BTN_TRIGGER_HAPPY35
#define X52_BTN_PG_DN       BTN_TRIGGER_HAPPY36     //BTN_TRIGGER_HAPPY36
#define X52_BTN_UP          BTN_TRIGGER_HAPPY37     //BTN_TRIGGER_HAPPY37
#define X52_BTN_DN          BTN_TRIGGER_HAPPY38     //BTN_TRIGGER_HAPPY38
#define X52_BTN_MFD_SELECT  BTN_TRIGGER_HAPPY39     //BTN_TRIGGER_HAPPY39
```
ive commented your original order and used the order the system has by default

MNS26, 2023-01-03:
> Uh-oh, that doesn't sound right at all. Could you paste the output of `x52bugreport`? I'm assuming you have the latest release, i.e. 0.3.0

where can i find it?

nirenjan, 2023-01-03:
You should just be able to run `x52bugreport`.

MNS26, 2023-01-03:
nope

MNS26, 2023-01-03:
nvm

nirenjan, 2023-01-03:
What version are you running?

MNS26, 2023-01-03:
with or without the modified kernel module loaded

MNS26, 2023-01-03:
ill do both

MNS26, 2023-01-03:
without the kernel module loaded
```sh
> x52bugreport
libx52 bugreport
================
Package version: 0.3.0
Build version: v0.3.0
Built on: 2023-01-03T16:56:02+0100
version-info $Id: 9ff692993dd94efda1a444da79e175cf111e631b $

Built against:
==============
libusb API version: 0x01000109
hidapi version: 0.12.0

System versions:
================
libusb: 1.0.26.11724 (http://libusb.info)
hidapi: 0.12.0

System info:
============
Linux 6.1.2-zen1-1-zen x86_64 (#1 ZEN SMP PREEMPT_DYNAMIC Sat, 31 Dec 2022 17:40:33 +0000)

Device info:
============
Connection failure
~/github/asusctl main !111 ?3 >                                                                                                                                                                                                                                    
```

With the kernel module loaded (modified button layout)
```sh
> x52bugreport
libx52 bugreport
================
Package version: 0.3.0
Build version: v0.3.0
Built on: 2023-01-03T16:56:02+0100
version-info $Id: 9ff692993dd94efda1a444da79e175cf111e631b $

Built against:
==============
libusb API version: 0x01000109
hidapi version: 0.12.0

System versions:
================
libusb: 1.0.26.11724 (http://libusb.info)
hidapi: 0.12.0

System info:
============
Linux 6.1.2-zen1-1-zen x86_64 (#1 ZEN SMP PREEMPT_DYNAMIC Sat, 31 Dec 2022 17:40:33 +0000)

Device info:
============
Connection failure
~/github/asusctl main !111 ?3 >      
```

nirenjan, 2023-01-03:
Could you please rerun it with `sudo`?

MNS26, 2023-01-03:
with module loaded
```sh
> sudo x52bugreport
[sudo] password for noah: 
libx52 bugreport
================
Package version: 0.3.0
Build version: v0.3.0
Built on: 2023-01-03T16:56:02+0100
version-info $Id: 9ff692993dd94efda1a444da79e175cf111e631b $

Built against:
==============
libusb API version: 0x01000109
hidapi version: 0.12.0

System versions:
================
libusb: 1.0.26.11724 (http://libusb.info)
hidapi: 0.12.0

System info:
============
Linux 6.1.2-zen1-1-zen x86_64 (#1 ZEN SMP PREEMPT_DYNAMIC Sat, 31 Dec 2022 17:40:33 +0000)

Device info:
============
Device ID: vendor 0x06a3 product 0x0762 version 0x2000
Device name: 'Logitech' 'X52 Professional H.O.T.A.S.'
Serial number: '(null)'
```

without the module
```sh
> sudo x52bugreport
libx52 bugreport
================
Package version: 0.3.0
Build version: v0.3.0
Built on: 2023-01-03T16:56:02+0100
version-info $Id: 9ff692993dd94efda1a444da79e175cf111e631b $

Built against:
==============
libusb API version: 0x01000109
hidapi version: 0.12.0

System versions:
================
libusb: 1.0.26.11724 (http://libusb.info)
hidapi: 0.12.0

System info:
============
Linux 6.1.2-zen1-1-zen x86_64 (#1 ZEN SMP PREEMPT_DYNAMIC Sat, 31 Dec 2022 17:40:33 +0000)

Device info:
============
Device ID: vendor 0x06a3 product 0x0762 version 0x2000
Device name: 'Logitech' 'X52 Professional H.O.T.A.S.'
Serial number: '(null)'
```

nirenjan, 2023-01-03:
Could you also run `sudo usbhid-dump -m 06a3:0762`

MNS26, 2023-01-03:
```sh
> sudo usbhid-dump -m 06a3:0762
[sudo] password for noah: 
001:022:000:DESCRIPTOR         1672771511.937843
 05 01 09 04 A1 01 09 01 A1 00 09 30 09 31 15 00
 26 FF 03 75 0A 95 02 81 02 75 02 95 01 81 01 09
 35 15 00 26 FF 03 75 0A 95 01 81 02 09 32 09 33
 09 34 09 36 15 00 26 FF 00 75 08 95 04 81 02 05
 09 19 01 29 27 15 00 25 01 95 27 75 01 81 02 75
 05 95 01 81 01 05 01 09 39 15 01 25 08 35 00 46
 3B 01 66 14 00 75 04 95 01 81 42 05 05 09 24 09
 26 15 00 25 0F 75 04 95 02 81 02 C0 C0
 ```

nirenjan, 2023-01-03:
That is rather weird, it should have worked fine without the kernel module - I'm wondering if there's a bug in the kernel itself.

MNS26, 2023-01-03:
hmm  i have no experience on that area

MNS26, 2023-01-03:
> That is rather weird, it should have worked fine without the kernel module - I'm wondering if there's a bug in the kernel itself.

is there anything i can do to test that?

nirenjan, 2023-01-03:
I can rewrite the kernel module to send the exact same events as the stock kernel - that should help isolate some of this, as well as ensure that the end user doesn't have to switch between events

nirenjan, 2023-01-03:
Can you try to use the kernel module in the `update-lkm` branch? This should work exactly like the without module case, except that the report parsing is done by the module and not by the default kernel USB HID layer.

nirenjan, 2023-01-03:
I don't have my joystick handy, so I can't test it myself at the moment.

MNS26, 2023-01-03:
it works

MNS26, 2023-01-03:
it shows axis 7 and 8 but doesnt go nuts

MNS26, 2023-01-03:
something that i dont understand is that it shows all my inputs double... but it did that without module as well

[33 more replies]

closed 2023-01-07 by MNS26