nirenjan/libx52 #49: kernel module repeatedly resets after some time

MNS26, 2023-01-08:
**Describe the bug**
after inserting the `saitek_x52.ko` module to fix the mouse axis showing in proton
it causes the controller to randomly reset on button inputs (afaik only from the joystick) after a good while, its not a fix time after adding sadly.

**To Reproduce**
Steps to reproduce the behavior:
1. insert `saitek_x52.ko` module
2. keep the controller connected (havent been able to determine the time)
3. pressing (ringed) hat Up or for me also the pinky trigger + main trigger causes reset

**Expected behavior**
it not to reset

**Screenshots/Logs**
* If applicable, add screenshots to help explain your problem.
* Attach the detailed X52 daemon logs.
logs dont show anything out of the ordinary, journalctl -f only show it resetting usb due to EMI? (even tho it plugged in directly) 
https://user-images.githubusercontent.com/53867268/211157287-d1565122-3d8f-4e93-8299-a67c2796a0d1.mp4
https://user-images.githubusercontent.com/53867268/211122692-9bdb0dea-ed53-4e94-a8c2-ab2807551653.mp4

**Environment (please complete the following information):**
 - Output of `x52bugreport`
```sh
> sudo !!
sudo x52bugreport
libx52 bugreport
================
Package version: 0.3.0
Build version: v0.3.0
Built on: 2023-01-03T16:56:02+0100
version-info $Id: 9ff692993dd94efda1a444da79e175cf111e631b $

Built against:
==============
libusb API version: 0x01000109
hidapi version: 0.12.0

System versions:
================
libusb: 1.0.26.11724 (http://libusb.info)
hidapi: 0.13.0

System info:
============
Linux 6.1.3-zen1-1-zen x86_64 (#1 ZEN SMP PREEMPT_DYNAMIC Wed, 04 Jan 2023 16:28:17 +0000)

Device info:
============
Device ID: vendor 0x06a3 product 0x0762 version 0x2000
Device name: 'Logitech' 'X52 Professional H.O.T.A.S.'
Serial number: '(null)'
~/github/libx52/kernel_module master !2 ?4 >   
```
**Additional context**

without the kernel module it works fine, the only issue is the system seeing axis 7 & 8 which are the mouse X & Y and causing proton to go nuts

nirenjan, 2023-01-08:
My concern here is why it's showing up with only the custom kernel module and not otherwise. The module doesn't do anything besides parse the raw events, and inject events. There may be changes in newer kernels that could be causing this to be incompatible with the new kernel, but I haven't been keeping track of the kernel changes for a while now.

For proton, I'm wondering if it's an issue with handling ABS_MISC and ABS_MISC+1. I can look into a custom module that just changes those to use ABS_TILT_X and ABS_TILT_Y, maybe that would help? I'm not sure though.

MNS26, 2023-01-08:
if you mean with the change i made to it.... it also happens with the one in the repo.
if you mean compared to without... idk.

MNS26, 2023-01-08:
> My concern here is why it's showing up with only the custom kernel module and not otherwise. The module doesn't do anything besides parse the raw events, and inject events. There may be changes in [...]

its only proton that is angry about it, wine doesnt go nuts actually. (but has hell performance in most games)

MNS26, 2023-01-08:
ayy its no more there 🥳 ...
installed system updates and the issue is gone

nirenjan, 2023-01-08:
That may just be a kernel issue then - I haven't tested against the latest kernel, and have no intention of doing so.

closed 2023-01-14 by MNS26