nirenjan/libx52 #50: journal flooded with usb IO errors

MNS26, 2023-01-14:
**Describe the bug**
when plugging in the X52 the Journal is flooded with 
`ERROR: x52d_device.c:179 Error 8 when updating X52 device: USB input/output error`</r>
`ERROR: x52d_device.c:179 Error 15 when updating X52 device: Pipe error` </r>
and a occasional `ERROR: x52d_io.c:75 Error 5 reading from X52 I/O device: I/O error`

**To Reproduce**
Steps to reproduce the behavior:

1. plug in X52 pro HOTAS

**Expected behavior**
It not screaming errors left and right

**Screenshots/Logs**
[image]
[x52d Journal log](https://github.com/nirenjan/libx52/files/10418731/x52d.log)

**Environment (please complete the following information):**
 - Output of `x52bugreport`
```sh
> sudo x52bugreport
libx52 bugreport
================
Package version: 0.3.0
Build version: v0.3.0
Built on: 2023-01-03T16:56:02+0100
version-info $Id: 9ff692993dd94efda1a444da79e175cf111e631b $

Built against:
==============
libusb API version: 0x01000109
hidapi version: 0.12.0

System versions:
================
libusb: 1.0.26.11724 (http://libusb.info)
hidapi: 0.13.1

System info:
============
Linux 6.1.5-zen2-1-zen x86_64 (#1 ZEN SMP PREEMPT_DYNAMIC Thu, 12 Jan 2023 22:42:30 +0000)

Device info:
============
Device ID: vendor 0x06a3 product 0x0762 version 0x2000
Device name: 'Logitech' 'X52 Professional H.O.T.A.S.'
Serial number: '(null)'
```

**Additional context**
Add any other context about the problem here.

nirenjan, 2023-01-15:
This seems rather weird, could you try plugging it into a different USB port and let me know if you still see the error?

MNS26, 2023-01-15:
hmm it still happens

MNS26, 2023-01-15:
hmm compared a USB packet capture i now with back when i 1st got the X52 and noticed it changed USB endpoints</r>
`new capture (left) | old capture (right)`
[image]

MNS26, 2023-01-17:
ive been looking through the `x52_core.c` together with `io_core.c` to try and figure out where its calling accessing the interface from and seeing if its possible to add a for loop to go through all the interfaces the usb has and filter out the HID one

nirenjan, 2023-01-21:
Could you please provide the output of `lsusb -vvv -d 06a3:`? The data that I have for the X52 Pro indicates that there should only be endpoint 0x81.

nirenjan, 2023-01-21:
Please also provide the output of `sudo usbhid-dump -d 06a3`

MNS26, 2023-01-25:
sorry had been VERY busy lately... once im home i will send the output (in a couple hours)

MNS26, 2023-01-25:
well sh#t came in between... i wont be able to do it for the next few days sadly

nirenjan, 2023-01-25:
No problem, I'm also busy with other stuff, so whenever you can get to it. I suspect that Logitech may have changed something after they acquired Saitek.

nirenjan, 2023-01-30:
That's certainly a possibility, but I'd like to take a look at the reports anyway.

nirenjan, 2023-01-30:
One more approach we could use is to use the following command on a relatively new kernel.

```
find /sys/devices -ipath '*:06a3:*/report_descriptor' -exec sh -c 'echo {} ; od -t x1 -Anone {}' \;
```

MNS26, 2023-02-05:
i found some free time.
`sudo lsusb -vvv -d 06a3:`
```sh

Bus 001 Device 011: ID 06a3:0762 Saitek PLC Saitek X52 Pro Flight Control System
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x06a3 Saitek PLC
  idProduct          0x0762 Saitek X52 Pro Flight Control System
  bcdDevice           20.00
  iManufacturer           1 Logitech
  iProduct                2 X52 Professional H.O.T.A.S.
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     124
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```
`sudo usbhid-dump -d 06a3`
```sh
001:016:000:Failed to retrieve report descriptor: Pipe error
```
`find /sys/devices -ipath '*:06a3:*/report_descriptor' -exec sh -c 'echo {} ; od -t x1 -Anone {}' \;`
returns nothing

MNS26, 2023-02-05:
nvm thats whe wrong one

MNS26, 2023-02-05:
there we go... (forgot i had a emulated device connected)

i found some free time.
`sudo lsusb -vvv -d 06a3:`
```sh
Bus 001 Device 043: ID 06a3:0762 Saitek PLC Saitek X52 Pro Flight Control System
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x06a3 Saitek PLC
  idProduct          0x0762 Saitek X52 Pro Flight Control System
  bcdDevice           20.00
  iManufacturer           1 Logitech
  iProduct                2 X52 Professional H.O.T.A.S.
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 
      iInterface              4 TinyUSB Serial
      CDC Header:
        bcdCDC               1.20
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
[... 75 lines cut ...]
  (Bus Powered)
```
`sudo usbhid-dump -d 06a3`
```sh
001:043:002:DESCRIPTOR         1675594742.595625
 05 01 09 04 A1 01 09 01 A1 00 09 30 09 31 15 00
 26 FF 03 75 0A 95 02 81 02 75 02 95 01 81 01 09
 35 15 00 26 FF 03 75 0A 95 01 81 02 09 32 09 33
 09 34 09 36 15 00 25 FF 75 08 95 04 81 02 05 09
 19 01 29 27 15 00 25 01 95 27 75 01 81 02 75 05
 95 01 81 00 05 01 09 39 15 01 25 08 35 00 46 3B
 01 66 14 00 95 01 75 04 81 42 05 05 09 24 09 26
 15 00 25 0F 95 04 75 02 81 02 C0 C0
```
`find /sys/devices -ipath '*:06a3:*/report_descriptor' -exec sh -c 'echo {} ; od -t x1 -Anone {}' \;`
```sh
 05 01 09 04 a1 01 09 01 a1 00 09 30 09 31 15 00
 26 ff 03 75 0a 95 02 81 02 75 02 95 01 81 01 09
 35 15 00 26 ff 03 75 0a 95 01 81 02 09 32 09 33
 09 34 09 36 15 00 25 ff 75 08 95 04 81 02 05 09
 19 01 29 27 15 00 25 01 95 27 75 01 81 02 75 05
 95 01 81 00 05 01 09 39 15 01 25 08 35 00 46 3b
 01 66 14 00 95 01 75 04 81 42 05 05 09 24 09 26
 15 00 25 0f 95 04 75 02 81 02 c0 c0
```

nirenjan, 2023-02-05:
So, I had a look against an older report descriptor, it looks like there are some differences.

1. The reports for `Z`, `Rx`, `Ry` and `Slider` have the logical maximum set to `-1` instead of `255`
2. The portion of the descriptor for the hat report indicates 4 entries of 2 bits each instead of just 2 entries of 4 bits each.

See the following for the older descriptor format.

```
 05 01 09 04 A1 01 09 01 A1 00 09 30 09 31 15 00
 26 FF 03 75 0A 95 02 81 02 75 02 95 01 81 01 09
 35 15 00 26 FF 03 75 0A 95 01 81 02 09 32 09 33
 09 34 09 36 15 00 26 FF 00 75 08 95 04 81 02 05
 09 19 01 29 27 15 00 25 01 95 27 75 01 81 02 75
 05 95 01 81 01 05 01 09 39 15 01 25 08 35 00 46
 3B 01 66 14 00 75 04 95 01 81 42 05 05 09 24 09
 26 15 00 25 0F 75 04 95 02 81 02 C0 C0
```

You can decode the descriptor by going to https://eleccelerator.com/usbdescreqparser/

nirenjan, 2023-02-05:
Since you mentioned you were playing with the firmware, it's possible that you changed something, thereby causing issues with the driver. One thing I noted is that your device is including a USB serial port and 3 interfaces (including 1 HID), whereas my older Saitek one has only a single HID interface. 

I would need to get a new X52 pro from Logitech and debug the USB packets with the Windows driver to find out what changed, or if it has really changed.

MNS26, 2023-02-07:
hmm i might be able to shuffle around the HID and serial port... but it would mean ripping apart like half the firmware (yikes)

MNS26, 2023-02-07:
i managed to restore the device descriptor as best as i can... there are _some_ differences sadly but i hope they arent critical </r>

(Left) recreated | (Right) original
[image]

MNS26, 2023-02-07:
i couldnt get `bDeviceClass`, `bDeviceSubClass`, `bDeviceProtocol`, `bMaxPacketSize0`, `iSerial`, but what im most concerned about is the `wMaxPacketSize` difference

MNS26, 2023-02-07:
welp... F*@# 
[image]
it still doesnt seem to work.... (is there any data being sent back to the daemon? Could it be it somehow not sending it?)

nirenjan, 2023-02-07:
The X52 has a protocol to update the MFD and LEDs, that portion doesn't seem to be working, and it may be due to the firmware itself. You can see the protocol documentation [here](https://github.com/nirenjan/libx52/blob/master/docs/specs/vendor_api.md) and the implementation [here](https://github.com/nirenjan/libx52/blob/master/libx52/x52_control.c#L69-L101).

One thing you could do is to use the same device on a Windows system (a VM will do fine as long as you can pass through the device) with the official drivers, and check if they work as expected. If those do work, then there is indeed a change that Logitech has made, and I would need to get a capture of the packets sent between the Windows driver and the device in order to implement the relevant features.

With that said, I would like to know how you modified the firmware and if there is something that you could share.

nirenjan, 2023-02-07:
From my, admittedly limited, knowledge of USB internals, there are several descriptors, of which, the device descriptor is only one. You need to update the config descriptor, interface descriptor and endpoint descriptor(s) as well. In addition to that, there would be some logic in the firmware that handles the vendor API, whether that's coming through the serial port or the control endpoint is unknown at this time.

MNS26, 2023-02-15:
i managed to track down the original firmware and reflashed it... seems to have fixed it

nirenjan, 2023-02-18:
OK, looks like this was a firmware issue. I'll go ahead and close this issue.

closed 2023-02-18 by nirenjan