nirenjan/libx52 #52: AUR package FTBFS (failed to build from source)

trinitronx, 2024-06-02:
**Describe the bug**

The `PKGBUILD` currently on AUR failed to build with compilation errors in `pinelog.c`.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `yay -Sy libx52`
3. Attempt to build the package
4. See error in build log below

**Expected behavior**
Package should build and install successfully.

**Screenshots/Logs**
 - Build log error:

       make[3]: Entering directory '/home/exampleuser/.cache/yay/libx52/src/libx52-0.3.0/build/lib/pinelog'
       /bin/sh ../../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../../../lib/pinelog -I../..   -D_FORTIFY_SOURCE=3 -O2 -DPINELOG_SHOW_LEVEL=1 -DPINELOG_SHOW_BACKTRACE=1 -DPINELOG_BUFFER_SZ=1024 -fno-strict-aliasing -Wnested-externs -Wmissing-prototypes -Wstrict-prototypes -Wdeclaration-after-statement -Wimplicit-function-declaration -Wold-style-definition -Wjump-misses-init -Wall -Wextra -Wundef -Wwrite-strings -Wpointer-arith -Wmissing-declarations -Wredundant-decls -Wno-unused-parameter -Wno-missing-field-initializers -Wformat=2 -Wcast-align -Wformat-nonliteral -Wformat-security -Wsign-compare -Wstrict-aliasing -Wshadow -Winline -Wmissing-format-attribute -Wmissing-noreturn -Winit-self -Wmissing-include-dirs -Wunused-but-set-variable -Warray-bounds -Wreturn-type -Wswitch-enum -Wswitch-default -Wduplicated-cond -Wduplicated-branches -Wlogical-op -Wrestrict -Wnull-dereference -Wdouble-promotion -Werror -Wno-suggest-attribute=format -Wno-error=unused-parameter -Wno-error=missing-field-initializers -I ../.. -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -MT libpinelog_la-pinelog.lo -MD -MP -MF .deps/libpinelog_la-pinelog.Tpo -c -o libpinelog_la-pinelog.lo `test -f 'pinelog.c' || echo '../../../lib/pinelog/'`pinelog.c
       libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../../lib/pinelog -I../.. -D_FORTIFY_SOURCE=3 -O2 -DPINELOG_SHOW_LEVEL=1 -DPINELOG_SHOW_BACKTRACE=1 -DPINELOG_BUFFER_SZ=1024 -fno-strict-aliasing -Wnested-externs -Wmissing-prototypes -Wstrict-prototypes -Wdeclaration-after-statement -Wimplicit-function-declaration -Wold-style-definition -Wjump-misses-init -Wall -Wextra -Wundef -Wwrite-strings -Wpointer-arith -Wmissing-declarations -Wredundant-decls -Wno-unused-parameter -Wno-missing-field-initializers -Wformat=2 -Wcast-align -Wformat-nonliteral -Wformat-security -Wsign-compare -Wstrict-aliasing -Wshadow -Winline -Wmissing-format-attribute -Wmissing-noreturn -Winit-self -Wmissing-include-dirs -Wunused-but-set-variable -Warray-bounds -Wreturn-type -Wswitch-enum -Wswitch-default -Wduplicated-cond -Wduplicated-branches -Wlogical-op -Wrestrict -Wnull-dereference -Wdouble-promotion -Werror -Wno-suggest-attribute=format -Wno-error=unused-parameter -Wno-error=missing-field-initializers -I ../.. -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -MT libpinelog_la-pinelog.lo -MD -MP -MF .deps/libpinelog_la-pinelog.Tpo -c ../../../lib/pinelog/pinelog.c  -fPIC -DPIC -o .libs/libpinelog_la-pinelog.o
       ../../../lib/pinelog/pinelog.c: In function ‘pinelog_init’:
       ../../../lib/pinelog/pinelog.c:123:33: error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
         123 |     module_level = calloc(sizeof(*module_level), count);
             |                                 ^
       ../../../lib/pinelog/pinelog.c:123:33: note: earlier argument should specify number of elements, later size of each element
       ../../../lib/pinelog/pinelog.c:129:32: error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
         129 |     module_name = calloc(sizeof(*module_name), count);
             |                                ^
       ../../../lib/pinelog/pinelog.c:129:32: note: earlier argument should specify number of elements, later size of each element
       cc1: all warnings being treated as errors
       make[3]: *** [Makefile:1541: libpinelog_la-pinelog.lo] Error 1
       make[3]: Leaving directory '/home/exampleuser/.cache/yay/libx52/src/libx52-0.3.0/build/lib/pinelog'
       make[2]: *** [Makefile:427: all-recursive] Error 1
       make[2]: Leaving directory '/home/exampleuser/.cache/yay/libx52/src/libx52-0.3.0/build/lib'
       make[1]: *** [Makefile:2964: all-recursive] Error 1
       make[1]: Leaving directory '/home/exampleuser/.cache/yay/libx52/src/libx52-0.3.0/build'
       make: *** [Makefile:1339: all] Error 2
       ==> ERROR: A failure occurred in build().
           Aborting...
        -> error making: libx52-exit status 4

<details><summary>Full <code>yay -Sy libx52</code> Log</summary><p>

```
$ yay -Sy libx52
 
[sudo] password for exampleuser: 
:: Synchronizing package databases...
 manjaro-sway                               30.4 KiB   198 KiB/s 00:00 [---------------------------------------] 100%
 mesa-nonfree is up to date
 core is up to date
 extra is up to date
 multilib is up to date
AUR Explicit (1): libx52-0.3.0-1
Sync Make Dependency (2): cmocka-1.1.7-1, doxygen-1.11.0-1
:: PKGBUILD up to date, skipping download: libx52
  1 libx52                           (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> A
:: Deleting (1/1): /home/exampleuser/.cache/yay/libx52
HEAD is now at 9069ad5 Update PKGBUILD to backup x52d.conf
warning: could not open directory 'pkg/': Permission denied
Removing libx52_0.3.0.orig.tar.gz
Removing pkg/
Removing src/
  1 libx52                           (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> A
 -> libx52: No changes -- skipping
[... 299 lines cut ...]
:: Processing package changes...
(1/2) removing doxygen                                                 [---------------------------------------] 100%
(2/2) removing cmocka                                                  [---------------------------------------] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) 
 -> Failed to install the following packages. Manual intervention is required:
libx52 - exit status 4

```

</p></details> 

**Environment (please complete the following information):**
 - Output of `x52bugreport`: `N/A` - could not build, so this utility is not installed
 - `gcc --version`:

       gcc (GCC) 14.1.1 20240522
       Copyright (C) 2024 Free Software Foundation, Inc.
       This is free software; see the source for copying conditions.  There is NO
       warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
       

**Additional context**
N/A

nirenjan, 2024-06-04:
Thank you for the bug report, it seems like newer versions of GCC are flagging this and raising a warning, which is treated as an error due to the `-Werror` flag.

closed 2024-06-04 by nirenjan

trinitronx, 2024-06-05:
> Thank you for the bug report, it seems like newer versions of GCC are flagging this and raising a warning, which is treated as an error due to the `-Werror` flag.

Thanks! :pray: :tada: :clap:

I've been able to recompile from latest `master` git branch, and tested by creating a [pre-release `PKGBUILD`](https://gitlab.archlinux.org/trinitronx/libx52).  So far as I can tell, everything's working!

nirenjan, 2024-06-05:
Thank you. I'm no longer using Arch - would you be interested in maintaining the PKGBUILD on AUR?

trinitronx, 2024-06-07:
> Thank you. I'm no longer using Arch - would you be interested in maintaining the PKGBUILD on AUR?

Sure!  I'm currently using Manjaro on my daily driver PC, and am able to maintain the `PKGBUILD`.

nirenjan, 2024-06-07:
Ah, I'm not a "Trusted User"/Package maintainer on Arch, but I did add you as a co-maintainer for the libx52 repo on AUR.

trinitronx, 2024-06-08:
> My understanding was that anybody could contribute to AUR, though you'd have to be a trusted user to contribute to the main repository.

Oh, that makes sense!  I must have been misunderstanding the Wiki.

>  I did add you as a co-maintainer for the libx52 repo on AUR. 

:pray: Thanks! I'm happy to help out with this `PKGBUILD` :grinning: 

Should I now be able to update the `PKGBUILD` on AUR by simply pushing to the SSH Git repo URL?  (e.g. `ssh://aur@aur.archlinux.org/libx52.git`).

Also, it looks like the `source` URL in the `libx52` AUR package currently set to the GitHub release tarball URL: `https://github.com/nirenjan/x52pro-linux/releases/download/v${pkgver}/${pkgname}_${pkgver}.orig.tar.gz`.  My testing pre-release `PKGBUILD` is using `source` from Git master branch `HEAD`.  Although, due to most AUR guidelines using Git repo source from `master` / `main` branches should be reserved for packages with the `-git` suffix.   That is to say in other words:  A separate package `libx52-git` would use Git source, meanwhile `libx52` package would use a typical distributed source tarball... like from Autotools `make dist` or `make distcheck` targets.  Likewise, `libx52-bin` would be the package name for a `PKGBUILD` that just distributes a pre-compiled upstream binary (as in most closed-source or proprietary things, or just software that only distributes a `.deb` or `.rpm` or some other foreign package rather than source code).

If I'm going to update to latest `make dist` release tarball, it would need to be uploaded to GitHub releases.  I noticed that there is a [`release` GitHub workflow][1] to do this... Is this still functional?

[1]: https://github.com/nirenjan/libx52/blob/master/.github/workflows/release.yml

nirenjan, 2024-06-08:
I believe you should be able to push to the AUR repo, but I've never done this before, do let me know if I need to do something else. 

The release workflow runs when a new tag is pushed to the repo. I'll do so once I'm back home.

nirenjan, 2024-06-09:
While the actions that the release workflow have been deprecated, they still work (for now). I've created release v0.3.1, so you should be able to update the PKGBUILD to use that new version.

trinitronx, 2024-06-10:
Ok, thanks!

I just tried doing a version bump and running the command to update sha256 checksums for the `PKGBUILD` (`makepkg --geninteg`), but it failed to download the `.tar.gz` file.

I checked into what's going on, and it looks like the URL has a version mismatch for the release version and version in filename:

```
https://github.com/nirenjan/libx52/releases/download/v0.3.1/libx52_0.3.0.orig.tar.gz
                                                      ^^^ here     ^^^ and here
```

I figured that it's probably a bug since it looks like the [`Version` file still has `0.3.0`].  So, I can hold off and wait until that gets fixed rather than fudge the versions in the `PKGBUILD` `source` URL.

[1]: https://github.com/nirenjan/libx52/blob/master/Version#L1

nirenjan, 2024-06-10:
D'oh! Thanks for catching it... That's what I get when I'm trying to make a release when I really should be in bed. Please use v0.3.2 instead.

trinitronx, 2024-06-12:
> Please use v0.3.2 instead.

Ok, great!  I've found some time to do this today.  `PKGBUILD` for Version `0.3.2-1` is now pushed and [live on the AUR][1]! 

[1]: https://aur.archlinux.org/packages/libx52

commit 21050e40 (2024-06-04): Fix syntax of calloc calls in pinelog.c

`calloc` requires the count to be the first argument, and the size
parameter to be the second argument. However, this has not really caused
issues in the past, and older compilers were not so strict about it.

However, newer compilers (at least GCC 14) triggers a warning on this
and causes the build to fail.

Fixes #52